seafloor-geodesy/gnatss #83: Add reference time

johnbdesanto, 2023-06-22:
When converting from time in seconds to a date-time string, the program needs to know what time the seconds are referenced to in order to complete the operation accurately. As an example, it is common for seconds to be referenced to UNIX time, which is Jan 1 1970. However, certain GPS processing software may have different reference times. For instance, the records produced by the JPL software GipsyX references seconds to 12:00 on Jan 1 2000.

Thus, we should consider implementing a flag so the user can set a time reference for the conversion so that input data can be read correctly

lsetiawan, 2023-06-23:
Thanks for raising this issue! This is a really great description. 😄

johnbdesanto, 2023-06-23:
I want to leave some more thoughts on this issue.

This will likely vary by surface platform, but the data feeds most likely to have their own independent time stamps are the two-way travel times, the roll/pitch/heading values, and the GPS data. Of these, the time reference of the GPS data will depend on the software used to process the data, but I would expect the others to be relatively static.

What I'm considering is whether it is worth adding up to three variables for input, whether it is fine to stick to one, whether it may be better to rely on the user to make this transformation themselves and provide the relevant references in the documentation, or whether it may be simplest to write a small and flexible utility for the end user to use.