seafloor-geodesy/gnatss #85: Upgrade the use of pydantic version 1 to 2

lsetiawan, 2023-07-18:


mansenfranzen, 2023-07-24:
You can now test the new **autodoc_pydantic** version supporting pydantic v2 via `pip install autodoc_pydantic==2.0.0`. The dev docs already showcase v2 support (see [here](https://autodoc-pydantic.readthedocs.io/en/main-2.x/index.html)). Migration guide is available [here](https://autodoc-pydantic.readthedocs.io/en/main-2.x/users/faq.html#migration-guide-from-v1-to-v2).

Any feedback is more than appreciated. The current **autodoc_pydantic** version `2.0.0` on pypi remains yanked for one week until final release.

lsetiawan, 2023-07-25:
Thanks @mansenfranzen for letting me know! I really appreciate your hardwork in maintaining the `autodoc_pydantic` library! 🍾

PR #133 (2023-08-16): refactor: update codebase to use Pydantic v2
## Overview

This PR refactor the pydantic models to use Pydantic version 2 and its best practices. Additionally, the version for `autodoc_pydantic` is now bumped to 2 as well.

Closes #85
closed 2023-08-16 by lsetiawan