seafloor-geodesy/gnatss #122: 🚨 Need tests for `loaders` module

lsetiawan, 2023-08-09:
## Overview

Currently the [`loaders`](https://github.com/uw-ssec/offshore-geodesy/blob/main/src/gnatss/loaders.py) module is at around [22% code coverage](https://app.codecov.io/gh/uw-ssec/offshore-geodesy/blob/main/src%2Fgnatss%2Floaders.py), which means that there are a lot of parts of the code currently untested. Tests are needed for the following functions:

- [ ] Add test for `load_configuration`
- [ ] Add test for `load_sound_speed`
- [ ] Add test for `load_travel_times`
- [ ] Add test for `load_gps_solutions`
- [ ] Add test for `load_deletions`

lsetiawan, 2023-09-29:
@madhavmk Please refer to https://github.com/uw-ssec/offshore-geodesy/pull/158 for test data for the loaders.

madhavmk, 2023-11-07:
Closing this Issue as all sub-tasks are completed.

closed 2023-11-07 by madhavmk