seafloor-geodesy/gnatss #127: Add test for `load_deletions`

lsetiawan, 2023-08-09:


madhavmk, 2023-10-16:
Hello @lsetiawan,
I would require a sample `tests/data/output/deletions.csv` file in order to unit-testing this function.

lsetiawan, 2023-10-16:
You should be able to start this testing by loading in a `file_path` at `https://github.com/uw-ssec/offshore-geodesy/blob/main/tests/data/2022/NCL1/deletns.dat`. I think it's better if you can generate the deletions csv file rather than from a direct sample since this is something that is not from the data provider. The file is currently very simple. The columns names are `constants.DEL_STARTTIME` and `constants.DEL_ENDTIME`, and the values are j2k values. I suggest the following steps:

### Deletions file
1. Pick random travel times from `pxp_tt` file.
2. Create deletions csv to the path provided
3. Load in the deletions file and make sure that it's the same values you picked

### Outliers file
1. Pick random travel times from `pxp_tt` file
2. Create outliers csv to the path provided... the column you need is only `constants.TT_TIME`
3. When you run the functions, if you have both the outliers and deletions, they should be merged together so ensure that the values exists for both deletions and outliers

closed 2023-11-07 by lsetiawan