seafloor-geodesy/gnatss #130: Cross check time conversion with GipsyX

lsetiawan, 2023-08-10:


lsetiawan, 2023-08-10:
@johnbdesanto please see the notebook above for the expected conversion values and cross check against GipsyX. Let me know if you have any questions.

johnbdesanto, 2023-08-10:
Hi Don, I got the following:

(base) [jdesanto@ernie ~]$ sec2date 712215480
2022-07-27 17:38:00
(base) [jdesanto@ernie ~]$ sec2date 712215495
2022-07-27 17:38:15
(base) [jdesanto@ernie ~]$ sec2date 712215510
2022-07-27 17:38:30
(base) [jdesanto@ernie ~]$ sec2date 712215525
2022-07-27 17:38:45
(base) [jdesanto@ernie ~]$ date2sec 2022-07-27 17:38:00
712215480.0
(base) [jdesanto@ernie ~]$ date2sec 2022-07-27 17:38:15
712215495.0
(base) [jdesanto@ernie ~]$ date2sec 2022-07-27 17:38:30
712215510.0
(base) [jdesanto@ernie ~]$ date2sec 2022-07-27 17:38:45
712215525.0
(base) [jdesanto@ernie ~]$ sec2date 0
2000-01-01 12:00:00

It looks like everything that you posted in the notebook matches what we were using before.

lsetiawan, 2023-08-11:
Okay. Then I think it's safe to say that GipsyX is using Terrestrial Time Scale for this conversion. So we'll assume that for the input data. Do you agree?

lsetiawan, 2023-08-14:
There's really only one definition of J2000 and GipsyX is using this.

From wikipedia:
> J2000: One commonly used ECI frame is defined with the Earth's Mean Equator and Mean Equinox (MEME) at 12:00 [Terrestrial Time](https://en.wikipedia.org/wiki/Terrestrial_Time) on 1 January 2000. It can be referred to as J2K, J2000 or EME2000. The x-axis is aligned with the mean vernal equinox. The z-axis is aligned with the [Earth's rotation axis](https://en.wikipedia.org/wiki/North_Pole) (or equivalently, the [celestial North Pole](https://en.wikipedia.org/wiki/Celestial_pole)) as it was at that time. The y-axis is rotated by 90° East about the celestial equator.[[5]](https://en.wikipedia.org/wiki/Earth-centered_inertial#cite_note-5)

closed 2023-08-14 by lsetiawan