seafloor-geodesy/gnatss #149: Add a way to compute e,n,u components from residuals

lsetiawan, 2023-08-25:


lsetiawan, 2023-08-25:
@johnbdesanto could you paste here:
1. a final solution xyz
2. array center coordinates used
3. enu computed
4. the azimuth angle computed
5. the look angle computed

If you can point me to a resource on how you computed these azimuth and look angles that would be really great! Additionally, the formula for residuals (cm) -> e,n,u components. Thank you so much!

johnbdesanto, 2023-08-28:
The formula you posted are correct. You are also right in getting rid of the factor of 2, which is there because I assumed the residuals were two way. We have since shown that not to be the case, so it should be removed.

johnbdesanto, 2023-08-28:
For calculating the angles, say that t_n, t_e, and t_d are the north, east, and depth (positive down) of the transponder in meters and c_n, c_e are the north and east of the array center, also in meters (assume 0 depth for the center since it's at sea level). Then,

az = ATAN[ (t_n-c_n)/(t_e-c_e) ], alternatively ATAN2[ (t_n-cn), (t_e-c_e) ]
el = ATAN[ SQRT[ (t_n-c_n)^2 + (t_e-c_e)^2 ]/t_d ], alternatively ATAN2[ SQRT[ (t_n-c_n)^2 + (t_e-c_e)^2 ], t_d ]

There is no real reference for this, just basic trig that I drew out. In the past, I have calculated t_n, t_e, c_n, c_e by converting the a priori coordinates to UTM. However, with the python tools we have we should be able to directly compute t_n-c_n and t_e-c_e by converting the transponder a priori position to ENU coordinates with the array center as the origin.

johnbdesanto, 2023-08-28:
For a sanity check, consider NCL1, whose transponders and center have the following a priori:

     0    08       45.302064471    -124.978181346    -1176.5866     0.200000d0       1480.767
     1    12       45.295207747    -124.958752845    -1146.5881     0.320000d0       1480.816
     2    22       45.309643593    -124.959348875    -1133.7305     0.440000d0       1480.850
45.3023d0 -124.9656d0   ! Latitude/Longitude array center (decimal degrees)

For this array, I calculated the following azimuth and look angles for the transponders:

#Fetch 08 Azi Angle: -179.9, Look Angle: 39.99
#Fetch 12 Azi Angle: -57.16, Look Angle: 39.74
#Fetch 22 Azi Angle: 57.61, Look Angle: 40.02

lsetiawan, 2023-09-29:
@madhavmk I'll share with you some implementation code so you can implement the routine.

lsetiawan, 2023-10-16:
Update: @madhavmk I just went ahead and implemented the routine and plotting code for this on PR #174.

closed 2023-10-19 by lsetiawan

madhavmk, 2023-10-20:
Thanks for the PR pointers @lsetiawan . I am taking a look at them.