seafloor-geodesy/gnatss #175: Install and test out the software

lsetiawan, 2023-10-19:


johnbdesanto, 2023-10-23:
I was able to successfully install gnatss using the provided instructions. Will run some tests tomorrow and Wednesday.

johnbdesanto, 2023-10-24:
Here is my first pass at running gnatss with the 2023 NCL1 survey (same site, different year). Seems successful, but matplotlib was not installed on the fresh conda environment when I did the pip install gnatss.

(gnatss) [jdesanto@geod-proc01-prd-a056 NCL1_2023]$ gnatss run --extract-dist-center --extract-process-dataset --qc --distance-limit 150 --residual-limit 10000
Loading configuration ...
Configuration loaded.
Gathering sound_speed at /wd4/GPSA_PROCESSING/Cascadia/2023/NCL1/ctd/CTD_NCL1_Ch_Mi
Gathering travel_times at /wd4/GPSA_PROCESSING/Cascadia/2023/NCL1/**/WG_*/pxp_tt
Gathering gps_solution at /wd4/GPSA_PROCESSING/Cascadia/2023/NCL1/**/posfilter/POS_FREED_TRANS_TWTT
Load sound speed profile data...
Computing harmonic mean...
lat=45.302064471 lon=-124.978181346 height=-1176.5866 internal_delay=0.2 sv_mean=1481.542 pxp_id='NCL1-1' azimuth=-178.48 elevation=39.99
lat=45.295207747 lon=-124.958752845 height=-1146.5881 internal_delay=0.32 sv_mean=1481.513 pxp_id='NCL1-2' azimuth=-55.73 elevation=39.75
lat=45.309643593 lon=-124.959348875 height=-1133.7305 internal_delay=0.44 sv_mean=1481.5 pxp_id='NCL1-3' azimuth=59.01 elevation=40.02
Finished computing harmonic mean
Load deletions data...
Load travel times...
Cleaning travel times data...
Load GPS data...
Cross referencing transmit, reply, and gps solutions...
Filtering out data outside of distance limit...
Preparing data inputs...
Perform solve...
--- 25877 epochs, 77631 measurements ---
After iteration: 1, rms residual = 274.18 cm, error factor = 65.512
NCL1-1
D_x = 1.763030e-02 m, Sigma(x) = 1.327666e-04 m
D_y = 6.206877e-02 m, Sigma(y) = 1.321839e-04 m
D_z = -1.783726e-02 m, Sigma(z) = 1.31018e-04 m
NCL1-2
D_x = 1.763030e-02 m, Sigma(x) = 1.327666e-04 m
D_y = 6.206877e-02 m, Sigma(y) = 1.321839e-04 m
D_z = -1.783726e-02 m, Sigma(z) = 1.31018e-04 m
NCL1-3
D_x = 1.763030e-02 m, Sigma(x) = 1.327666e-04 m
D_y = 6.206877e-02 m, Sigma(y) = 1.321839e-04 m
D_z = -1.783726e-02 m, Sigma(z) = 1.31018e-04 m
After iteration: 2, rms residual = 274.03 cm, error factor = 65.478
NCL1-1
D_x = -2.201396e-06 m, Sigma(x) = 1.327673e-04 m
D_y = 8.109158e-07 m, Sigma(y) = 1.32184e-04 m
D_z = 5.571606e-07 m, Sigma(z) = 1.310184e-04 m
NCL1-2
D_x = -2.201396e-06 m, Sigma(x) = 1.327673e-04 m
D_y = 8.109158e-07 m, Sigma(y) = 1.32184e-04 m
D_z = 5.571606e-07 m, Sigma(z) = 1.310184e-04 m
NCL1-3
D_x = -2.201396e-06 m, Sigma(x) = 1.327673e-04 m
D_y = 8.109158e-07 m, Sigma(y) = 1.32184e-04 m
D_z = 5.571606e-07 m, Sigma(z) = 1.310184e-04 m

---- FINAL SOLUTION ----
NCL1-3
x = -2574109.5264 +/- 1.327673e-04 m del_e = -0.0211 +/- 1.340284e-04 m
y = -3681766.2796 +/- 1.32184e-04 m del_n = 0.0308 +/- 1.352079e-04 m
z = 4510790.7136 +/- 1.310184e-04 m del_u = -0.0556 +/- 1.265733e-04 m
Lat. = 45.302064748083936 deg, Long. = -124.97818161553542, Hgt.msl = -1203.2321594597208 m
NCL1-3
x = -2574109.5264 +/- 1.327673e-04 m del_e = -0.0211 +/- 1.340282e-04 m
y = -3681766.2796 +/- 1.32184e-04 m del_n = 0.0308 +/- 1.352079e-04 m
z = 4510790.7136 +/- 1.310184e-04 m del_u = -0.0556 +/- 1.265735e-04 m
Lat. = 45.295208024068934 deg, Long. = -124.95875311423809, Hgt.msl = -1173.2336681832692 m
NCL1-3
x = -2574109.5264 +/- 1.327673e-04 m del_e = -0.0211 +/- 1.340282e-04 m
y = -3681766.2796 +/- 1.32184e-04 m del_n = 0.0308 +/- 1.352079e-04 m
z = 4510790.7136 +/- 1.310184e-04 m del_u = -0.0556 +/- 1.265735e-04 m
Lat. = 45.30964387019226 deg, Long. = -124.95934914431396, Hgt.msl = -1160.3760602712146 m
------------------------

There are 3 outliers found during this run.
This is 0.01% of the total number of data points.
Please re-run the program again to remove these outliers.

Saving the distance from center file to /wd4/GNATSS_tests/NCL1_2023/output/dist_center.csv
Saving the latest residuals to /wd4/GNATSS_tests/NCL1_2023/output/residuals.csv
Saving the latest residual outliers to /wd4/GNATSS_tests/NCL1_2023/output/outliers.csv
Saving the process results dataset to /wd4/GNATSS_tests/NCL1_2023/output/process_dataset.nc
Traceback (most recent call last):

  File "/home/jdesanto/miniconda3/envs/gnatss/bin/gnatss", line 8, in <module>
    sys.exit(app())

  File "/home/jdesanto/miniconda3/envs/gnatss/lib/python3.10/site-packages/gnatss/cli.py", line 122, in run
    from .ops.qc import plot_enu_comps, plot_residuals

  File "/home/jdesanto/miniconda3/envs/gnatss/lib/python3.10/site-packages/gnatss/ops/qc.py", line 3, in <module>
    import matplotlib.pyplot as plt

ModuleNotFoundError: No module named 'matplotlib'

johnbdesanto, 2023-10-24:
Note, the above was solved when I did a conda install matplotlib and re-logged into the machine via ssh -X. Here are the figures generated for an early run with residual outlier limit of 1000:

[image]

[image]

johnbdesanto, 2023-10-24:
Ok, I think that I've gleamed enough from this test (NCL1 2023 data) as I am going to. Some thoughts:

1) The runtime messages only report the final solution for the third transponder. Probably a typo somewhere.

```
---- FINAL SOLUTION ----
NCL1-3
x = -2574109.5029 +/- 1.360096e-04 m del_e = 0.007 +/- 1.372997e-04 m
y = -3681766.2951 +/- 1.354155e-04 m del_n = 0.0384 +/- 1.385156e-04 m
z = 4510790.7237 +/- 1.342222e-04 m del_u = -0.0489 +/- 1.296680e-04 m
Lat. = 45.30206481692833 deg, Long. = -124.97818125624207, Hgt.msl = -1203.225493334314 m
NCL1-3
x = -2574109.5029 +/- 1.360096e-04 m del_e = 0.0071 +/- 1.372995e-04 m
y = -3681766.2951 +/- 1.354155e-04 m del_n = 0.0384 +/- 1.385156e-04 m
z = 4510790.7237 +/- 1.342222e-04 m del_u = -0.0489 +/- 1.296682e-04 m
Lat. = 45.295208092859156 deg, Long. = -124.95875275498649, Hgt.msl = -1173.2269962538799 m
NCL1-3
x = -2574109.5029 +/- 1.360096e-04 m del_e = 0.0071 +/- 1.372995e-04 m
y = -3681766.2951 +/- 1.354155e-04 m del_n = 0.0384 +/- 1.385155e-04 m
z = 4510790.7237 +/- 1.342222e-04 m del_u = -0.0489 +/- 1.296682e-04 m
Lat. = 45.309643938968925 deg, Long. = -124.95934878497202, Hgt.msl = -1160.3693866216388 m
------------------------
```

2) The final solution is off by about 1.4 cm in the east component and 0.2 cm in the north component compared to what I was able to generate with the fortran code. The reason was because I did not flag as many residuals as I did with the old code. For the gnatss run, I went down to 50 cm before stopping when I could have gone down to ~40 before starting to flag good data. However, even with this there are still erroneous points well within the threshold. There are two recommendations I have that could help with this. First, we should look into implementing a threshold for east and north residuals since that is another avenue for data flagging. Second, we should get the deletions file to take times in something more akin to a datetime string instead of j2000 seconds since that is more amenable to manual flagging.

[image]

johnbdesanto, 2023-10-24:
When I mention flagging by east/north residual, I don't think it should be too hard since the east/north components are being calculated already and are paired with a timestamp. Basically, ask for a threshold (say 5 cm) and 

if ( abs(east/north) > threshold ) -> flag timestamp

johnbdesanto, 2023-10-24:
Another thought on flagging. I put together this plot very quickly based upon the residuals I posted in the above post. It shows the difference between the maximum residual and the minimum residual. Of note are the epochs where the difference is inflated, which correspond to the spikes in east-north residual and the poor residuals in the raw plot. You could look at this difference and set a flagging threshold of ~15-20 and get rid of many of the poor quality data points.

[image]

johnbdesanto, 2023-10-25:
Ok, I've run another test, this time with some legacy data collected with the old wave glider. I have chosen to try processing the 2017 survey collected at a different site in Cascadia, NNP1, to verify that the code still worked with a different site defined in the config.yaml file. The big difference with this data set is that the transducer wait time is 0.1 seconds and has not been corrected during processing. This means that the entries in the pxp_tt file, ex:

`25-JUL-17 12:30:00   2806272   2897177   3036648         0`

correspond to the following in the POS_FREED_TRANS_TWTT:

```
   554257800.1000000  -2629443.269      -3728285.669       4442205.495      0.2318898805E-03  0.1355408286E-07  -.9548982269E-07  0.1355408286E-07  0.2910790772E-03  0.7062928895E-07  -.9548982269E-07  0.7062928895E-07  0.3470023037E-03
   554257802.9062721  -2629441.524      -3728286.631       4442204.525      0.7670018681E-03  0.7228365037E-07  -.7372732053E-06  0.7228365037E-07  0.8247937343E-03  0.6375368044E-06  -.7372732053E-06  0.6375368044E-06  0.8815382791E-03
   554257802.9971770  -2629441.386      -3728286.590       4442204.556      0.8324168908E-03  0.8031686103E-07  -.8249094623E-06  0.8031686103E-07  0.8900087492E-03  0.7107877056E-06  -.8249094623E-06  0.7107877056E-06  0.9468553234E-03
   554257803.1366481  -2629441.224      -3728286.507       4442204.623      0.2650340658E-03  0.7555223771E-08  -.1196982051E-06  0.7555223771E-08  0.3257116932E-03  0.1633846776E-06  -.1196982051E-06  0.1633846776E-06  0.3832156082E-03
```

(Note the delay has been added to the POS_FREED_TRANS_TWTT epoch. And we were also kind of aware of this, but I cross-verified the epoch using python's datetime function and it lines up when you don't account for leap seconds.)

In short, the test failed with the following output:

```
(gnatss) [jdesanto@geod-proc01-prd-a056 NNP1_2017]$ gnatss run --extract-dist-center --extract-process-dataset --qc --distance-limit 150 --residual-limit 10000
Loading configuration ...
Configuration loaded.
Gathering sound_speed at /wd4/GPSA_PROCESSING/Cascadia/JdF_18/NNP1/ctd/CTD_NNP1_Ch_Mi
Gathering travel_times at /wd4/GPSA_PROCESSING/Cascadia/JdF_17/NNP1/**/WG_*/pxp_tt
Gathering gps_solution at /wd4/GPSA_PROCESSING/Cascadia/JdF_17/NNP1/**/posfilter/POS_FREED_TRANS_TWTT
Load sound speed profile data...
Computing harmonic mean...
lat=44.43958943 lon=-125.18595698 height=-1348.2479 internal_delay=0.2 sv_mean=1481.725 pxp_id='NNP1-1' azimuth=62.57 elevation=44.78
lat=44.418374435 lon=-125.185219222 height=-1363.2877 internal_delay=0.32 sv_mean=1481.754 pxp_id='NNP1-2' azimuth=-59.99 elevation=44.72
lat=44.429374842 lon=-125.210922477 height=-1379.914 internal_delay=0.44 sv_mean=1481.784 pxp_id='NNP1-3' azimuth=177.79 elevation=44.83
Finished computing harmonic mean
Load deletions data...
Load travel times...
Traceback (most recent call last):

  File "/home/jdesanto/miniconda3/envs/gnatss/bin/gnatss", line 8, in <module>
    sys.exit(app())

  File "/home/jdesanto/miniconda3/envs/gnatss/lib/python3.10/site-packages/gnatss/cli.py", line 83, in run
    _, _, resdf, dist_center_df, process_ds, outliers_df = main(

  File "/home/jdesanto/miniconda3/envs/gnatss/lib/python3.10/site-packages/gnatss/main.py", line 793, in main
    all_observations = load_data(all_files_dict, config)

  File "/home/jdesanto/miniconda3/envs/gnatss/lib/python3.10/site-packages/gnatss/main.py", line 507, in load_data
    all_travel_times = load_travel_times(

  File "/home/jdesanto/miniconda3/envs/gnatss/lib/python3.10/site-packages/gnatss/loaders.py", line 155, in load_travel_times
    all_travel_times[constants.TIME_ASTRO] = all_travel_times.apply(

  File "/home/jdesanto/miniconda3/envs/gnatss/lib/python3.10/site-packages/pandas/core/frame.py", line 10037, in apply
    return op.apply().__finalize__(self, method="apply")

  File "/home/jdesanto/miniconda3/envs/gnatss/lib/python3.10/site-packages/pandas/core/apply.py", line 837, in apply
    return self.apply_standard()

  File "/home/jdesanto/miniconda3/envs/gnatss/lib/python3.10/site-packages/pandas/core/apply.py", line 963, in apply_standard
    results, res_index = self.apply_series_generator()

  File "/home/jdesanto/miniconda3/envs/gnatss/lib/python3.10/site-packages/pandas/core/apply.py", line 979, in apply_series_generator
    results[i] = self.func(v, *self.args, **self.kwargs)

  File "/home/jdesanto/miniconda3/envs/gnatss/lib/python3.10/site-packages/gnatss/loaders.py", line 156, in <lambda>
    lambda row: AstroTime.strptime(

  File "/home/jdesanto/miniconda3/envs/gnatss/lib/python3.10/site-packages/astropy/time/core.py", line 1992, in strptime
    tt, fraction = _strptime._strptime(to_string(time), format_string)

  File "/home/jdesanto/miniconda3/envs/gnatss/lib/python3.10/site-packages/astropy/extern/_strptime.py", line 350, in _strptime
    raise ValueError("time data %r does not match format %r" %

ValueError: time data '25-jul-17 12:30:00' does not match format '%d-%b-%y %H:%M:%S.%f'
```

It seems the issue at play is the time cross-check, although with just the output to the terminal window I'm not sure if the wait time is the issue or the leap seconds are the issue.

johnbdesanto, 2023-10-25:
Looked into this a little more. Here is an example record from a pxp_tt file from the 2022 NCL1 survey, which we know worked:

```
(gnatss) [jdesanto@geod-proc01-prd-a056 NNP1_2017]$ head -1 ../../GPSA_PROCESSING/Cascadia/JdF_22/NCL1/209/WG_20220728/pxp_tt
28-JUL-22 00:00:11.00   2298282   2931785         0         0
```

What is immediately noticeable is that the timestamp for the 2022 data has a float second value while the 2017 data has an integer second. So my guess is now that the gnatss i/o is reading the entry and getting confused by finding a integer where it expects a float.

johnbdesanto, 2023-11-22:
I found another bug. If I run gnatss on a remote server that I have logged into with the ssh -Y command I get the following error:

```
Saving the distance from center file to /wd4/GNATSS_tests/NDP1_2022/output/dist_center.csv
Saving the latest residuals to /wd4/GNATSS_tests/NDP1_2022/output/residuals.csv
Saving the process results dataset to /wd4/GNATSS_tests/NDP1_2022/output/process_dataset.nc
qt.qpa.plugin: Could not load the Qt platform plugin "xcb" in "" even though it was found.
This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.

Available platform plugins are: eglfs, minimal, minimalegl, offscreen, vnc, webgl, xcb.

Aborted (core dumped)
```

Logging in with ssh -X fixes this.

johnbdesanto, 2023-11-22:
I have also identified a data set that I would like us to implement as a code benchmark. The data set is the 2022 survey of the Cascadia array NDP1. I have taken the liberty of creating a deletions file and have transferred that file to GNATSS.

The results of the original fortran scripts are:
```

----------------------------------------------------------------------------------
               FINAL SOLUTION FOR17759 EPOCHS, 53277 MEASUREMENTS:
----------------------------------------------------------------------------------
After iteration  2, rms residual =     21.79 cm, error factor =     5.216
 09 TRANSPONDER (17759 MEASUREMENTS):
   x =  -2604416.4042 +/- 1.61196D-04 m   del_e =      0.1806 +/- 1.60999D-04 m
   y =  -3705738.2206 +/- 1.58216D-04 m   del_n =     -0.0268 +/- 1.62177D-04 m
   z =   4472870.9024 +/- 1.56888D-04 m   del_v =         0.2056 +/- 1.52998D-04 m
   Lat. =  44.832681118 deg., Long. = -125.099792616 deg., Hgt.msl = -1832.6164 m
 10 TRANSPONDER (17759 MEASUREMENTS):
   x =  -2606819.0424 +/- 1.61196D-04 m   del_e =      0.1807 +/- 1.60998D-04 m
   y =  -3705463.8707 +/- 1.58216D-04 m   del_n =     -0.0267 +/- 1.62178D-04 m
   z =   4471710.7685 +/- 1.56888D-04 m   del_v =         0.2055 +/- 1.52999D-04 m
   Lat. =  44.817929409 deg., Long. = -125.126647165 deg., Hgt.msl = -1829.4395 m
 11 TRANSPONDER (17759 MEASUREMENTS):
   x =  -2606247.3505 +/- 1.61196D-04 m   del_e =      0.1807 +/- 1.60997D-04 m
   y =  -3703528.8255 +/- 1.58216D-04 m   del_n =     -0.0268 +/- 1.62179D-04 m
   z =   4473632.0917 +/- 1.56888D-04 m   del_v =         0.2055 +/- 1.52998D-04 m
   Lat. =  44.842324959 deg., Long. = -125.134817994 deg., Hgt.msl = -1830.5545 m
```

The results of GNATSS are: 

```
--- 17759 epochs, 53277 measurements ---
.
.
.
---- FINAL SOLUTION ----
NDP1-1
x = -2604416.4271 +/- 1.611955e-04 m del_e = 0.1805 +/- 1.609979e-04 m
y = -3705738.2529 +/- 1.582154e-04 m del_n = -0.0274 +/- 1.621764e-04 m
z = 4472870.941 +/- 1.568879e-04 m del_u = 0.2608 +/- 1.529981e-04 m
Lat. = 44.832681113201346 deg, Long. = -125.09979261690818, Hgt.msl = -1858.3711800229146 m
NDP1-2
x = -2606819.0652 +/- 1.611955e-04 m del_e = 0.1806 +/- 1.609969e-04 m
y = -3705463.9031 +/- 1.582154e-04 m del_n = -0.0273 +/- 1.621770e-04 m
z = 4471710.807 +/- 1.568879e-04 m del_u = 0.2608 +/- 1.529986e-04 m
Lat. = 44.817929404342145 deg, Long. = -125.12664716628078, Hgt.msl = -1855.1942329929761 m
NDP1-3
x = -2606247.3734 +/- 1.611955e-04 m del_e = 0.1806 +/- 1.609965e-04 m
y = -3703528.8578 +/- 1.582154e-04 m del_n = -0.0274 +/- 1.62178e-04 m
z = 4473632.1303 +/- 1.568879e-04 m del_u = 0.2607 +/- 1.529979e-04 m
Lat. = 44.84232495350724 deg, Long. = -125.1348179949484, Hgt.msl = -1856.3092628994827 m
------------------------
```

So the good news is that the del_e and del_n results are very close between the two runs, <1mm. However, the actual transponder coordinates differ by ~2-3 cm and the del_v estimate differs by ~6 cm. In addition, the Hgt.msl estimates differ by an amount that is roughly equal to the geoid undulation in the config.yaml file.

While the del_e and del_n are the most important results, we should investigate the others.

lsetiawan, 2023-11-27:
> While the del_e and del_n are the most important results, we should investigate the others.

@johnbdesanto,

It is great to see that you've found a bug on this front, though it's a bit concerning. I definitely agree that we should investigate to what's going on here. Is there a way for you to send me the necessary data for this calculation? Thanks!

lsetiawan, 2023-11-27:
> qt.qpa.plugin: Could not load the Qt platform plugin "xcb" in "" even though it was found.
This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.

This seems like something with the plotting I imagined. Thanks for catching this. I'll look into this issue.

lsetiawan, 2023-12-21:
The first release of the software can be found at https://github.com/seafloor-geodesy/gnatss/releases/tag/v0.1.0. Closing this issue since there will be another round of installing and testing in future releases.

closed 2023-12-21 by lsetiawan