seafloor-geodesy/gnatss #220: build: Need a config to ignore test data

lsetiawan, 2024-02-08:
@madhavmk brought up a point that currently test datasets might not be ignored in the built code... need a way to ignore these. One way is to use config with `Manifest.in` file to ignore.

madhavmk, 2024-02-08:
The point of concern is the large size of the `./tests` directory, which causes the `gnatss` source distribution tar file to be unnecessarily large. In the future we are likely to increase the `./tests` directory size significantly by adding additional data files. 
`
>[SSEC/offshore-geodesy]$ du -sh *
4.0K    LICENSE
0       MANIFEST.in
12K     README.md
844K    docs
4.0K    noxfile.py
4.0K    pyproject.toml
140K    src
15M     tests

>[SSEC/offshore-geodesy]$ python -m build
...
>[SSEC/offshore-geodesy]$ du -sh dist/*
44K     dist/gnatss-0.1.dev106-py3-none-any.whl
4.7M    dist/gnatss-0.1.dev106.tar.gz
`

After excluding the `./tests` directory from the build step, the source distribution file reduced in size significantly. This speeds up `gnatss` pip installation speed due to faster downloading, and lesser unzipping.
`
>[SSEC/offshore-geodesy]$ du -sh *
4.0K    LICENSE
0       MANIFEST.in
12K     README.md
844K    docs
4.0K    noxfile.py
4.0K    pyproject.toml
140K    src
15M     tests
>[SSEC/offshore-geodesy]$ python -m build
...
>[SSEC/offshore-geodesy]$ du -sh dist/*
44K     dist/gnatss-0.1.dev107-py3-none-any.whl
788K    dist/gnatss-0.1.dev107.tar.gz
`

madhavmk, 2024-02-08:
`MANIFEST.in` file is used only when using the `setuptools` build system. However as mentioned in the `pyproject.toml` we are using `hatchling` build system instead. We can therefore remove this MANIFEST file. 
[Reference](https://hatch.pypa.io/1.9/meta/faq/#build-system)

closed 2024-02-08 by lsetiawan