seafloor-geodesy/gnatss #226: feat: Create loaders for wave glider L1 data for velocity and roll-pitch-heading

lsetiawan, 2024-02-29:
## Overview

To mitigate any issues in repeating timestamp mentioned in #224, we need a way to simply read from an L1 data from the following files and respective file specification:
- NCL1_INSPVAA.dat: https://docs.novatel.com/OEM7/Content/SPAN_Logs/INSPVA.htm
- NCL1_INSSTDEVA.dat: https://docs.novatel.com/OEM7/Content/SPAN_Logs/INSSTDEV.htm

We'll need to parse the file, convert the time values, and pass through the data for covariance and data values of velocity as well as roll pitch heading.

madhavmk, 2024-03-20:
* Get it to a stage of DataFrame. One for std-dev, and one for data values.
* Merge one of the dataframes with pxp-tt
* Merge the other with pxp-tt
* After merging, then try sorting.

madhavmk, 2024-04-04:
Closed by PR #231

closed 2024-04-04 by madhavmk